nionio6915/Euclid_Probe #6: homexy.g

bilsch, 2022-03-21:
Hey mind sharing your homexy too?

nionio6915, 2022-03-22:
sure what part? 
; ***********************************************************
; homexy.g
; ***********************************************************
; 
if boards[0].vIn.current < 23
    M291 S2 R"Turn on power?" P"Power is OFF?" 
    M292 P0
    M80
    G4 P1000                           ; wait 1 second for power to stabilize
elif boards[0].vIn.current > 23
   M561                                 ; clear any bed transform
   M290 S0 R0                           ; clear baby steps 
   M564 S0 H0                           ; allow movement outside limits
                                        ; Home X and Y at the high end of the axis
   G91                                  ; relative positioning
   G1 H2 Z5 F6000                       ; lift Z relative to current position
   G1 H1 X292 Y325 F9000                ; move quickly to X and Y axis endstops and stop there (first pass)
   G1 H2 X-5 Y-5   F9000                ; go back a few mm
   G1 H1 X292 Y325 F360                 ; move slowly to X and Y axis endstops once more (second pass)
   G1 H2 Z-5 F600                       ; lift Z relative to current position
   G90                                  ; absolute positioning
   M564 S1

nionio6915, 2022-03-22:
i tihnk you want this one too - 
```
; ***********************************************************
; homeall.g
; ***********************************************************
; called to home all axes
M98 P"0:/sys/homexy.g"                         ; home x y
M98 P"0:/sys/homez.g"                           ; home z by calling submacros

```

bilsch, 2022-03-22:
you rock thanks!

bilsch, 2022-03-22:
oh, so boards[0] is the duet board right? Then we can index into other boards if we want? say I have a/b on the 3hc and its board 1 - boards[1]... would get that detail for the other board right?

nionio6915, 2022-03-22:
I do not know the Duet3x hardware at all.