nionio6915/Euclid_Probe #7: Config for Klipper 00-euclid_exampleV4.cfg can't be parsed by clipper

VTopoliuk, 2022-05-05:
Klipper version: v0.10.0-387
Error:
```
Error loading template 'gcode_macro DEPLOY_PROBE:gcode': TemplateSyntaxError: expected token 'end of print statement', got 'DOCK'
```
Happens without any modification to original file

nionio6915, 2022-06-05:
can you provide your printer.cfg and the macro file you are using?

arikhris, 2022-06-05:
`# array variables implementation and 
# macro setups credited to user yolodubstep
# see  https://github.com/blalor/vcore3-ratos-config 
# for updates and details
#  __________________________________________________________________________
#  |                                                                        |
#  |                                                                        |
#  |                                                                        |
#  |                                                                        |
#  |                                                                        |
#  |                                                                        |
#  |                                                                        |
#  |                                                                        |
#  |                                                                        |
#  |                                                                        |
#  |                                * Probe Ready Position                  |
#  |                                  X150 Y150                             |
#  |                                                                        |
#  |                                                                        |
#  | * Dock Re-entry staging  position                                      |
#  |   X0 Y70                                                               |
#  |                                                                        |
#  |                                                                        |
#  | * Dock Exit Position                                                   | 
#  |   X0 Y40                                                               |
#  |                                                                        |
#  |                                                                        |
#  |                                                                        |
#  |                                                                        |
#  |                                                                        |
#  |   X0 Y0    X30 Y0       X100 Y0                                        |
#  | * Dock   * Dock Side  * Dock Preflight                                 |
#  |________________________________________________________________________| 
#
# Above is example 300x300 bed to coorelate with macros and movements below.
# This example is for a fixed dock, fixed gantry/carraige and moving bed motion
# system like RailCore, Ender5, V-Core3, etc...
# Delta printes will be similar
# Moving gantry printers like Voron need a a few tweaks to ensure clearances
#  - see cpmments in the code

[gcode_macro EuclidProbe]
description: config vars for Euclid probe deploy/stow

## Replace the cordinates to suit your printer for deployment steps
variable_position_preflight: [ 100, 348 ] # positions for probe to have clear path to dock 
variable_position_side:      [  30, 348 ] # position for probe near dock to swipe on 
variable_position_dock:      [   0, 348 ] # dock position
## exit/re-entry staging
variable_position_exit:      [   0, 40 ] # exit position

## how much space to put between the bed and nozzle for homing
# RatOS
variable_bed_clearance: 15

# 300 * 60
variable_move_speeds: 18000

variable_batch_mode_enabled: False
variable_probe_state: None

gcode:
    RESPOND TYPE=command MSG="{ printer['gcode_macro EuclidProbe'] }"

# the following is a hardware probe config for the hardware configuration. 
# it can appear in the printer.cfg or in the euclid.cfg 
# USE IT IN ONE LOCATION ONLY

[homing_override]
# homing overide is machine specific. this example os for fixed gantry moving bed printer
[... 322 lines cut ...]
#    Z_TILT_ADJUST_ORIG
#    STOW_PROBE

# Macro to perform a QGL by wrapping it in DEPLOY_PROBE/STOW_PROBE macros
[gcode_macro QUAD_GANTRY_LEVEL]
rename_existing: QUAD_GANTRY_LEVEL_ORIGINIAL
#klipper
# you mane need to increase the horizontal_move_z: 15
gcode:

    DEPLOY_PROBE
    QUAD_GANTRY_LEVEL_ORIGINIAL
    STOW_PROBE

[gcode_macro PROBE_CALIBRATE]
rename_existing: PROBE_CALIBRATE_ORIG
gcode:
    {% set euclid_probe = printer["gcode_macro EuclidProbe"] %}  # this is the contents of line 144
    DEPLOY_PROBE
    G90
    G0 X{ printer.toolhead.axis_maximum.x/2 } Y{ printer.toolhead.axis_maximum.y/2 } F{ euclid_probe.move_speeds }
    PROBE_CALIBRATE_ORIG
    STOW_PROBE

`

arikhris, 2022-06-05:
same errror as op

nionio6915, 2022-06-06:
please upload your printer.cfg and your klipper/moonraker versions

nionio6915, 2022-06-07:
OK, we think we have a handle on this. 

Lines 189-190 are comments to provide users instructions to add a move command to clear the dock height on moves. I beleive that klipper is parsing over the comment and trying to evaluate whats inside the brackets. 

```
; fixed bed dock and moving gantry printers need to add a move command here to lower the gantry to dock height
; G0 Z{INSERT DOCK HEIGHT} F500
```

Could you to try the following-

1. After line 55 add a comment and a variable 
```
# dock clearance height for docking and stow macros
variable_dock_height: 15
```

2. Revise line 190 to 
```; G0 Z {euclid_probe.dock_height} F500```

I believe that this will elimintate the error.  I ask that you try it because my klipper machines are not gantries.