nionio6915/Euclid_Probe #16: Bug in 00-euclid_exampleV5.cfg crashes Bed into Hotend

MartinGiersberg, 2022-08-15:
When using the Euclide probe on a V-Core 3 - 400 Machine while running RatOS and having a Z-Offset >=+5mm the Code in the line 136 of "00-euclid_exampleV5.cfg", G0 Z5, will make the Bed move in the wron direction after homing Z. If the Z-Offset is <5mm this wont happen and the bed will move as expected after homing.

FIX: Change the value for Z in the line 136 so that is is always strictly larger than your Z-Offset

nionio6915, 2022-08-15:
thank your for catching this. There should have been a G91 before this. after some consideration, it has been revised to be the bed_clearance value. 

    # after 'G28 Z' the probe stays in contact with the bed; move it away bed_clearance height.
    G0 Z{euclid_probe.bed_clearance} F500

closed 2022-08-15 by nionio6915

MartinGiersberg, 2022-08-16:
I tested this code on my machine and it works like a charm. Thanks a lot!